Chr157i4n/PyTmcStepper #1: Schematics

sandric, 2022-01-07:
Hi, could you please provide schematics? I'm using tmc2209 with raspberry pi zero w, and this is what I get:

```
pi@pi:~/TMC2209_Raspberry_Pi $ python3 test_script_01_uart_connection.py 
---
SCRIPT START
---
TMC2209: Init
TMC2209: GPIO Init finished
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: after 10 tries not valid answer. exiting
TMC2209: Deinit
TMC2209: Motor output active: False
```

Would love to check if I had some error wiring, I just connected tmc2209 RX to RPi's TX(8 pin) and tmc2209 TX to RPi's RX(10 pin).

sandric, 2022-01-07:
Sorry, just realized that all works if I connect TX->TX and RX->RX straight, for some reason I thought it needs to be reversed. But schematics would also be very appeciated, since I also get from examples that apart from just uart it still needs 3-pin connection for movement, right?

Chr157i4n, 2022-01-07:
Hey,
yes apart the uart connection, you have the 3-pin connection (En, Dir, Step). Those three pins needs to be defined when you create the TMC_2209 object:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/aa63474ced9841be7515f99ab602589ce85a37a8/test_script_01_uart_connection.py#L17
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/aa63474ced9841be7515f99ab602589ce85a37a8/TMC_2209/TMC_2209_StepperDriver.py#L75-L82

sandric, 2022-01-07:
Yes, I already tested it and its working, you are right. What I'm not quite understand is why its needing both uart and 3pin? I mean I can control stepper with just 3 pins - I tested it without uart with simple c script. I thought that uart can be used instead of 3 pins simply sending commands to tmc mcu, to remove number of connections?

Chr157i4n, 2022-01-07:
You need the 3 pin connection (En, Dir, Step) on the TMC2209 when you want to drive the motor a given amount of steps.
The UART connection can be used to change setting in the driver, like the motor current or the microstep resolution.

I think with the TMC5160 you can drive the Motor completly over the communication interface (in that case SPI), because the TMC5160 has an inbuilt Motion Controller. The TMC2209 does not have one.

The TMC2209 has only an inbuilt "Step Pulse Generator":
"Some applications do not require a precisely co-ordinate motion – the motor just is required to move
for a certain time and at a certain velocity. The TMC2209 comes with an internal pulse generator for
these applications: Just provide the velocity via UART interface to move the motor" [from the TMC2209 Datasheet]

sandric, 2022-01-07:
@Chr157i4n Thanks a lot for explanation, clarified a lot! I'm not sure if issue should be closed though since it was originally about schematics, close it if you think its redundant.

commit 030a0db5 (2022-01-07): added schematics
closes #1
closed 2022-01-07 by Chr157i4n