Chr157i4n/PyTmcStepper #3: Can I use two TMC2209 with one Raspberry Pi board with this lib?

AndriiKnyr, 2022-01-28:


Chr157i4n, 2022-01-28:
Hello,
the problem with using two TMC2209 on one Raspberry Pi is the UART-Interface.
So driving both drivers/motor with the Step/Dir Pins is possible.
But change setting in two different drivers via UART is currently not supported in this libary.
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_datasheet_rev1.07.pdf
It is mentioned in Chapter 4.3 and 4.4:
[image]
It should be possible to use different addresses for the different drivers.
But currently that is not implementet.

igoro00, 2022-03-17:
self.mtr_id (in __init__ in TMC_2209_uart.py) is the slave address. It's 0 (or 0b00) by default. If you change it to 0b11 (or 3) it'll only talk to the driver that has both MS1 and MS2 set high. 

I'm using my stepper with this pcb that has screw terminals and dip switches etc.
[image]
I just ordered 2 more to test this out but in theory all I have to do is to make mtr_id configurable and create two more objects with different addresses.

closed 2022-03-17 by Chr157i4n

shaunmulligan, 2022-05-27:
@igoro00 did you ever manage to get this working? I have two drivers hooked up sharing the same serial connection but can't seem to get them both working simultaneously :(

Chr157i4n, 2022-05-27:
I tested it and you should be able to set the register setting via UART in both drivers and move the motor successively.

For the test script [test_script_06_multiple_drivers.py](https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/main/tests/test_script_06_multiple_drivers.py) i am using the address 0 (nothing connected to MS1 and MS2) and the address 1 for the second driver (Vcc connected to MS1 and nothing connected to MS2)

Can you share some more information about your setup?

see also:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/commit/eb1732ffe05665a0700b7c2550399e7e553790ec

shaunmulligan, 2022-05-28:
Thanks @Chr157i4n , I actually realised I had misunderstood the data sheet and that I needed a separate STEP and DIR line for each motor, so I now have it working correctly 👍

commit c17fe10c (2022-03-17): added address to print output
- closes #3