Chr157i4n/PyTmcStepper #4: Add to pip

igoro00, 2022-03-15:
It would be great if I could download this library from pip and keep it up to date without copying the code from the repo.

Current method works great for running tests but I want to build something more serious.

Chr157i4n, 2022-03-16:
Hey,

@Plexiable has already changed the structure to a Python PIP package:
https://github.com/Plexiable/TMC2209_Raspberry_Pi

i have now uploaded it to PyPI:
https://pypi.org/project/TMC-2209-Raspberry-Pi

could you please test whether everything works and give me a feedback?

you can install it with pip:
```
pip install TMC-2209-Raspberry-Pi
```
or
```
pip3 install TMC-2209-Raspberry-Pi
```

Chr157i4n, 2022-03-26:
@igoro00
if it doesn't work, feel free to open this issue again

closed 2022-03-26 by Chr157i4n