Chr157i4n/PyTmcStepper #25: Standstill current reduction not working (iHOLD)

SirMighty, 2022-07-13:
**Describe the bug**
When setting IHOLD and IRUN the current doesn't get reduced when the motor is standing still

**To Reproduce**
Set a different hold_current_multiplier with `tmc.setCurrent(600, hold_current_multiplier=0)`
When the stepper is enabled and standing still, it remains powered.

**Expected behavior**
When hold_current_multiplier=0, the stepper should freewheel (or spinning almost freely) when it is standing still and enabled

**Cause**
According to official documentation, when using uart, you should set `pdn_disable` in GCONF to 1 to enable standstill current reduction.
(see Documentation p.20)

**Solution**
Set pdn_disable to 1

Temporary fix:
``` python
gconf = tmc.readGCONF()
gconf = gconf | 1 << 6
tmc.tmc_uart.write_reg_check(0x00, gconf)
```

Chr157i4n, 2022-07-16:
i have added a function setPDNdisable.
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/923d4ca89d006f7f7c118a7a43fa25def31100be/src/TMC_2209/TMC_2209_StepperDriver.py#L623-L631

this function is beeing called in the setCurrent function unless you set "pdn_disable = False" in the setCurrent call.
So by default this should now disable PDN on the UART pin and therefore the current should get reduced at standstill.

Chr157i4n, 2022-07-20:
I will close this issue now. If you still have problems with the current reduction feel free to open this issue again

commit 923d4ca8 (2022-07-16): Standstill current reduction
- added setPDNdisable function
- setPDNdisable is called from setCurrent function
- with PDN disabled current should be correctly reduced at standstill
- fixes #25
closed 2022-07-20 by Chr157i4n