Chr157i4n/PyTmcStepper #27: TMC2209: after 10 tries not valid write access

SirMighty, 2022-07-28:
**Describe the bug**
After writing quite a few values to the TMC registers, the error occurs

**To Reproduce**
Write a setting in a loop, after a while the error occurs

**Error source**
The error occurs in TMC_2209_uart.write_reg_check() when the register IFCNT reaches 255. According to the Datasheet, the register IFCNT wraps around from 255 to 0, but that condition isn't handled, so it just throws an error, even though everyting is fine.

**Temporary Fix**
```python
    def write_reg_check(self, register, val, tries=10):
        ifcnt1 = self.read_int(reg.IFCNT)
        
        if(ifcnt1 == 255): # <-- wraparound fix
            ifcnt1 = -1

        while(True):
            self.write_reg(register, val)
            tries -= 1
            ifcnt2 = self.read_int(reg.IFCNT)
            if(ifcnt1 >= ifcnt2):
                print("TMC2209: writing not successful!")
                print("TMC2209: ifcnt:",ifcnt1,ifcnt2)
            else:
                return True
            if(tries<=0):
                print("TMC2209: after 10 tries not valid write access")
                self.handle_error()
                return -1
```
It would probably be prettier to add the right check to the error checking if instead of my "dirty" fix 😅

Chr157i4n, 2022-07-29:
Hey,
good job on finding that bug. And thanks for looking it up in the datasheet.

> It would probably be prettier to add the right check to the error checking if instead of my "dirty" fix

What exactly do you suggest?

In my opinion your solution is good.

SirMighty, 2022-08-01:
Well, you're right. I thought of integrating it into the existing if, but that would make the code unneccessarily complicated.

Thx for implementing the fixes so fast, appreciate it!

commit bc045e04 (2022-07-29): fix ifcnt wraps around from 255 to 0
fixes #27
closed 2022-08-01 by SirMighty