Chr157i4n/PyTmcStepper #28: Adding a function for continous turning

Taav00, 2022-08-03:
Hi, 

For a project I'm working on I needed to make a stepper turn continuously for up to an hour. I've used your library with VACTUAL to make it work by adding a function set to true the deinitalize flag at the end so the driver doesn't deinitalize itself at the end of the code. 

```
#-----------------------------------------------------------------------
# set deinitialize to true
#-----------------------------------------------------------------------       
    def setDeinitializeTrue(self):
        self._deinit_finished = True
```

Though this might be useful to someone else in an other project someday. Could you add it to the TMC_2209_StepperDriver.py ?

Chr157i4n, 2022-08-03:
Hey,

how do you stop the motor then after that one hour?

Cant you use something like this to move the motor for an hour:
```
tmc.setVActual(1000)
time.sleep(3600)
tmc.setVActual(0)
```

Taav00, 2022-08-04:
Hi,

The stepper driver is controlled through a function call to set a speed and after it has completed others tasks, the function is called again to set the speed to zero. 

This has the advantage of just setting the speed once and letting the Raspberry Pi do something else while the motor is turning continuously.

Chr157i4n, 2022-08-04:
i am currently trying to change the code according to the python styleguides. So i named the function `set_deinitialize_true`

commit 1c921c4f (2022-08-04): added set_deinitialize_true function
closes #28