Chr157i4n/PyTmcStepper #31: stallguard / finding values

rmalchow, 2022-10-31:
hi christian,

i have been using a bigtreetech board with marlin before. with the stallguard autohoming function of marlin, i was able to very smoothly find the endstops of my sliders. one light touch and stallguard got triggered. 

now, i wanted to try to do this sligthly differently - and i found your library. most things work quite nicely, but the one thing that seems difficult is finding threshholds for stallguard. sometimes, despite being stuck, the driver doesnt register this, while at other times, it triggers in situations where it shouldn't ...

i've been playing with various combinations of current limit and threshold values, but was unable to find anything reliable. 

do you have any recommendations / strategies for finding good settings? i'm using leadscrew sliders like this, which have rather clear end stops. 

[image]

regards

.rm

rmalchow, 2022-11-01:
hi christian,

awesome! this is really helpful. thank you very much. for my application - i was using a 3d printer board, but i only need one channel, and there's a raspi in the mix anyways - so it seems a waste to use a separate marlin board. i will try your suggestions, and hope i can lower my parts count :)

thanks!

.rm

Chr157i4n, 2022-11-10:
Hey,

i will close this issue now.
If you have still problems with tuning StallGuard, feel free to reopen this issue.

closed 2022-11-10 by Chr157i4n