Chr157i4n/PyTmcStepper #33: Do something else while running a stepper

arn13, 2023-02-09:
Hi
I am trying to build a machine that reads data from a serial port while simultaneously moving the reading-head of the serial device with a stepper motor.
Both operations, the reading & saving data and the movement of the motor work without any problems in their own scripts, but when I combine them with multiprocessing the stepper stutters and the Pi3B+ freezes up.
As I am a python-beginner I am not even sure if this is a problem with the library per se.

```
from TMC_2209.TMC_2209_StepperDriver import *
import time
import multiprocessing
import serial
import time
import csv

ser = serial.Serial('/dev/ttyACM0', 9600)
ser.flushInput()
print(ser.name) 

def reading():
    while True:
        try:
            ser_bytes = ser.readline()
            decoded_bytes = ser_bytes.decode('utf-8').rstrip()
            l=len(decoded_bytes)
            rdate=decoded_bytes[0:10]
            rtime=decoded_bytes[11:19]
            tagid=decoded_bytes[20:l-6]
            # removing the 5 last charaters for the temp with 2 decimals an seperating /t
            temp=decoded_bytes[l-5:l-1]
            # temperature with decimalpoint 2 decimals and leaving the unit
            print(rdate)
            print(rtime)
            print(tagid)
            print(temp)
            
            with open("test_data.csv","a") as f:
                writer = csv.writer(f,delimiter=',')
                writer.writerow([rdate,rtime,tagid,temp])
        except:
            print("Keyboard Interrupt")
            break

    

def movement():
    while True:
        tmc.set_max_speed_fullstep(500)
        tmc.run_to_position_steps(200)
        print("go to box1 start")
        tmc.set_max_speed_fullstep(10)
        tmc.run_to_position_steps(400)
        print("go to box1 end")

        tmc.set_max_speed_fullstep(500)
        tmc.run_to_position_steps(600)
        print("go to box2 start")
        
        tmc.set_max_speed_fullstep(10)
        tmc.run_to_position_steps(800)
        print("go to box1 end")
        

print("---")
print("SCRIPT START")
print("---")

#-----------------------------------------------------------------------
# initiate the TMC_2209 class
# use your pins for pin_en, pin_step, pin_dir here
#-----------------------------------------------------------------------
tmc = TMC_2209(21, 16, 20, loglevel=Loglevel.DEBUG)

#-----------------------------------------------------------------------
# set the loglevel of the libary (currently only printed)
# set whether the movement should be relative or absolute
# both optional
#-----------------------------------------------------------------------
tmc.set_loglevel(Loglevel.DEBUG)
tmc.set_movement_abs_rel(MovementAbsRel.ABSOLUTE)

#-----------------------------------------------------------------------
# these functions change settings in the TMC register
#-----------------------------------------------------------------------
tmc.set_direction_reg(False)
tmc.set_current(300)
tmc.set_interpolation(True)
tmc.set_spreadcycle(True)
tmc.set_microstepping_resolution(2)
tmc.set_internal_rsense(False)

print("---\n---")

#-----------------------------------------------------------------------
# these functions read and print the current settings in the TMC register
#-----------------------------------------------------------------------
tmc.readIOIN()
tmc.readCHOPCONF()
tmc.readDRVSTATUS()
tmc.readGCONF()

print("---\n---")

#-----------------------------------------------------------------------
# set the Acceleration and maximal Speed
#-----------------------------------------------------------------------
# tmc.set_acceleration(2000)
# tmc.set_max_speed(500)

#-----------------------------------------------------------------------
# set the Acceleration and maximal Speed in fullsteps
#-----------------------------------------------------------------------
tmc.set_acceleration_fullstep(500)
tmc.set_max_speed_fullstep(250)

#-----------------------------------------------------------------------
# activate the motor current output
#-----------------------------------------------------------------------
tmc.set_motor_enabled(True)

#-----------------------------------------------------------------------
# homing
# 1. param: DIAG pin
# 2. param: maximum number of revolutions. Can be negative for inverse direction
# 3. param(optional): StallGuard detection threshold
#-----------------------------------------------------------------------
print ("homing")
tmc.do_homing(26,-2000,50)

#then run moving and reading simultaneously

while True:
    print("moving and reading")
    if __name__ == '__main__':
        p1 = multiprocessing.Process(name='p1',target=movement)
        p = multiprocessing.Process(name='p',target=reading)
        p1.start()
        p.start()

#-----------------------------------------------------------------------
# deactivate the motor current output
#-----------------------------------------------------------------------
tmc.set_motor_enabled(False)

print("---\n---")

#-----------------------------------------------------------------------
# deinitiate the TMC_2209 class
#-----------------------------------------------------------------------
tmc.deinit()
del tmc

print("---")
print("SCRIPT FINISHED")
print("---")
```

Chr157i4n, 2023-02-09:
Hey,

in your main part you start both Threads in a while true loop:
```
while True:
    print("moving and reading")
    if __name__ == '__main__':
        p1 = multiprocessing.Process(name='p1',target=movement)
        p = multiprocessing.Process(name='p',target=reading)
        p1.start()
        p.start()
```
In both Thread functions there are while true loops aswell:
```
def reading():
    while True:
...
```

```
def movement():
    while True:
...
```
So both Threads dont start once, but get started infinitely often and never stop. So you have a high number of Threads running in parallel in a short time, which can freeze your RPi.

arn13, 2023-02-09:
Thank you for the quick reply, this partly solves the problem.
I removed the outer while loop, and inserted a timer to stop the disabling of the motor from happening instantly.

```
print("moving and reading")
    if __name__ == '__main__':
        p1 = multiprocessing.Process(name='p1',target=movement)
        p = multiprocessing.Process(name='p',target=reading)
        p1.start()
        p.start()

time.sleep(60)
```

Now the script works as expected.
For the first time it is being run that is.
If I run it again after it has ended or stopped the motor begins to stutter a bit (not as bad as before).
Also the data that gets read is gibberish on the second an following runs.
If I reboot the the Pi the first run after the reboot is fine again, but the problems with reruns are the same.
I suspect that the processes 'stick around' after the script has ended.
Sometimes even the motor continues running constantly in one direction.
Maybe some kind of cleanup is needed, but I have no idea yet how it should look like.

Chr157i4n, 2023-02-09:
Yes, remaining threads could be the cause for your behaviour.
Pls have a look at the documentation of the python multiprocessing library 
https://docs.python.org/3/library/multiprocessing.html

You need to stop your Threads at the end of your script.

1. The good way is to use the [join() function](https://docs.python.org/3/library/multiprocessing.html#multiprocessing.Process.join) of your thread in the main part to wait for the thread to finish. But then you need a stop condition inside your thread instead of the while true loop you currently have inside your threads.
something like this:
```
def reading(this_thread_should_still_run_xyz):
    while this_thread_should_still_run_xyz.value:
        ...

this_thread_should_still_run_xyz = Value('i', 1)
p = multiprocessing.Process(name='p',target=reading, args=(this_thread_should_still_run_xyz,))
p.start()

time.sleep(60)
this_thread_should_still_run_xyz.value = 0
p.join()
```
2.  The quick and dirty way would be to kill the threads after your time.sleep(60) with the [terminate() function](https://docs.python.org/3/library/multiprocessing.html#multiprocessing.Process.terminate) of the threads
```
p = multiprocessing.Process(name='p',target=reading)
p.start()

time.sleep(60)
p.terminate()
```

arn13, 2023-02-10:
The quick and dirty fix works!
I will have to read up a bit more and than I will try to stop the threads with join() for the proper implementation.
Thank you again for your help!

closed 2023-02-10 by arn13