Chr157i4n/PyTmcStepper #34: positions being skipped (due to performance issues)

arn13, 2023-03-15:
Hi, its me again.
After some more testing I am facing another problem:
Sometimes the stepper does not wait for one command to be fully carried out before starting the next. For example:
```
tmc.set_max_speed_fullstep(500)
tmc.run_to_position_steps(10)
        
tmc.set_max_speed_fullstep(100)
tmc.run_to_position_steps(2000)
```

This seems to be a performance problem because taxing the cpu (e.g. opening a browser) escalates the situation to the point where the movement is just a rapid back and forth.
Occasionally even the homing step in the beginning of my script gets skipped.
I tried tweaking the values for acceleration and speed, limiting them to 500 fullstep for acceleration and 250 fullstep for speed.

Do you have any suggestion on how tackle this problem?
Or is what I am trying to do more than what is possible in python.

I am running a Pi3B+ with raspbian 64bit btw. I could switch to a lite and or 32bit version if that helps.

Some background:
I just need the motor to run through a series of positions, preferably changing speeds between after every position. The idea is to have four "slow" sections and faster sections in between. At the end of the linear setup the contraption shall return to the first position.
the length of the whole rail measures about 12000 Steps

Chr157i4n, 2023-03-16:
Hey,
i am using a Pi4 and I did not encounter this error.
I have no clue why this occurs. So please test this more.
You could as an example put short pauses between each movement to further analyze this behaviour.

If your problem remains, please send me your whole script or attach it to your next message.

Chr157i4n, 2023-03-29:
Hey,
i would like to hear whether you found a solution,
otherwise i will close this issue in a few days.

arn13, 2023-03-29:
Sorry, I did not get to test this further.
The hardware is also in use at the moment.
(I switched to an Arduino for the motor control)
I will reopen the issue when I have gotten some test-results.

closed 2023-03-29 by arn13