Chr157i4n/PyTmcStepper #37: Stallguard fires right away

alxkos, 2023-11-05:
Getting another problem: 
Stallguard fires right away. 
only if i'm o very low speed and low acceleration. like 10,15
Here is my code:

`
tmc.setDirection_reg(False)
tmc.setVSense(True)

tmc.setIScaleAnalog(True)
tmc.setInterpolation(True)
tmc.setSpreadCycle(False)
tmc.setMicrosteppingResolution(1)
tmc.setInternalRSense(False)

tmc.setCurrent(200)
tmc.setAcceleration(100)
tmc.setMaxSpeed(100)

tmc.setMotorEnabled(True)

def my_callback(channel):  
    print("StallGuard!")
    tmc.stop()

tmc.getStallguard_Result()
tmc.setStallguard_Callback(21, 50, my_callback)

finishedsuccessfully = tmc.runToPositionSteps(2000, MovementAbsRel.relative)
tmc.getStallguard_Result()
if(finishedsuccessfully == True):
    print("Movement finished successfully")
else:
    print("Movement was not completed")

tmc.setMotorEnabled(False)

`

Simple setup, stepper 17hs13 with 0.4A
PSU is 9v
Any suggestions?
Badly wat to use Stallguard instead of using microswitches.

Thanks

**Versions**
- using esp32 
- OS Version windows
- Version of this library
- TMC2209 version and manufacturer dont have ver. aliexpress

Chr157i4n, 2023-11-05:
If Stallguard fires immediatly on starting the movement, try lowering your StallGuard threshold in the setStallguard_Callback function.

You can also use the doHoming2 function, which prints the StallGuard Results at the end for each step, to determine your threshold.

For me 50 as the threshold worked with your motor current and acceleration, but i have no load on my test setup.

alxkos, 2023-11-05:
Hey, thanks. I will check the homing function
Ive found the issue. its connects to speed, current, acceleration all together. 
want to find a "formula" to calculate it. But can continue working with it meanwhile

Chr157i4n, 2023-11-05:
Yes, that is correct. The driver measures the current flow to the motor and can therefore calculate the motor load:
[image]
[page 58 of the datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.08.pdf)

The motorload depends on the motor current, the acceleration and of course on what you drive with the motor. The datasheet also has a chapter (11.2 Tuning StallGuard4) which describes how you tune the threshold value.
Currently there is no easy way to see the stallguard result values while using the normal "runToPosition" function. Currently you can use the doHoming2 function for that.
I might implement a function for easier StallGuard tuning.

Chr157i4n, 2023-11-11:
in the last 2 releases i added the following to help with StallGuard tuning:

A function to monitor StallGuard result while moving the motor:
```python
test_stallguard_threshold(steps)
```

I added the proper calculation of the min_speed value for StallGuard:
```python
set_stallguard_callback(self, pin_stallguard, threshold, callback, min_speed = 100)
```
With the min_speed parameter you can now set a minimum speed in steps per second for StallGuard. Below that speed the DIAG will not be triggered. This should prevent StallGuard from triggering immediatly on accelerating.

and a wiki entry:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/wiki/StallGuard

Please give me feedback, whether this helps you, otherwise i will consider this issue resolved

closed 2023-11-18 by Chr157i4n