Chr157i4n/PyTmcStepper #44: Connecting two tmc2209 to serial

jasonthenderson, 2023-12-26:
What is the recommended way to connect multiple 2209 steppers to the same rpi uart?  Do I run them in serial or tie them both in to the same pin in parallel?

Chr157i4n, 2023-12-27:
The recommended way is to connnected all drivers in parallel to the RPi UART pin
[image]
[Datasheet page 20](https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.08.pdf)

How would you connect them in serial? Do you mean the two PDN_UART pins on some driver boards? They are there for compatibility reasons with different mainboards and often not both of them are connected.

jasonthenderson, 2023-12-27:
Hi Christian, thanks for the quick reply.  No, what I meant by hooking them in series was that on the PCB board I am designing, I have a line that runs from the Raspberry pi (actually Jetson nano) UART (Rx) to an 8 pin jack on the PCB, then from the pin on the PCB jack I have a trace that runs to the UART pin on the stepper chip one (I'm actually using the TMC2208) and then the trace continues on to the UART pin on the second TMC2208.  

[image]

Does it seem like that should work?

Chr157i4n, 2023-12-27:
Well, electrically speaking, your connection is still parallel and not serial. Just that the connection lines for each driver doesn't have the same length (which is not optimal). But for communication with only one driver at a time, it does not matter. So this should be fine (i am not a PCB expert ;) ).
Don't forget the 1kOhm resistor between RX and TX (i guess you have that already on your board below the image)

jasonthenderson, 2023-12-27:
Super, yes I have the 1k resistor between RX and TX, screenshot of that below.  

One other question - I was contemplating creating a trace in the PCB on one of the chips from MS1 to VIO in order to ensure that the chips always had different addresses.  Right now setting a different address for one of the chips would require a manual solder and wire connected between MS1 or MS2 and VIO on one of the chips.  My concern with doing this is that I may not use UART (right now I'm not in my code) and if I don't use UART I'm thinking that having that connection hardwired in the PCB would cause the chip with MS1 wired to VIO to instead do microstepping.  Am I correct in that thinking? I.e. I want my PCB to support both UART and non-UART method of driving the chips.
[image]

Chr157i4n, 2023-12-27:
[image]
[Datasheet page 9](https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.08.pdf)

If you put solder brigdes for VIO to MS1 and MS2, you can use that to either set the UART address or the microstepping resolution.
With a solder brigde it is not hardwired.

Chr157i4n, 2024-01-05:
I will close this issue now. If you still have problems with connecting multiple tmc2209 over serial feel free to open this issue again

closed 2024-01-05 by Chr157i4n