Chr157i4n/PyTmcStepper #45: about wiring diagram and wiring table

tom123478, 2024-01-03:
Thank you very much for sharing. I'm a little confused about the wiring diagram and wiring table and I'm hoping you can shed some light on it. In the wiring table, GND2 is connected to the GND of the Raspberry Pi. What does GND2 represent? My understanding is the GND of the 5V output. Why is the GND connected to the Raspberry Pi in the picture the power supply GND?

tom123478, 2024-01-03:
> ![grafik](https://private-user-images.githubusercontent.com/8201379/293860802-2637449b-7fc9-4a3c-9ba1-60dfc5fdb51b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyNzA2NTksIm5iZiI6MTcwNDI3MDM1OSwicGF0aCI6Ii84MjAxMzc5LzI5Mzg2MDgwMi0yNjM3NDQ5Yi03ZmM5LTRhM2MtOWJhMS02MGRmYzVmZGI1MWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDNUMDgyNTU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDAxYzcwOWQyYWI3ZmQxZGU0YWFkNzhhMDI3M2M0MjU2YWZkNDVmYzgwZDZiZDRiOTg4OWZjMTE1MThmYzcwMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.8yCp03__pd-MLGWn6ppwN5PMvkGLw6KU5glmsIWiFjc) https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf
> 
> Both GND pins are internally connected (with a GND plane i guess). So it does not matter to which GND pin you connect your supply GND and your Pi GND.
> 
> But yes, you're right, it's different in the table and in the picture.

I think I understand what you said, thank you very much

Chr157i4n, 2024-01-05:
I will close this issue now. If you still have problems with the wiring diagram or table feel free to open this issue again

commit 2e374829 (2024-01-03): renamed GND2 to GND
- switched order of wiring table
resolves #45
closed 2024-01-05 by Chr157i4n