Chr157i4n/PyTmcStepper #46: Compatible with Watterott TMC2209?

StBe121, 2024-01-13:
Hello, is your libary also working in UART mode with the Watterott TMC2209 I cannot get a working connection neither with the ESP32 nor the RPI, I followed the instructions to solder the pads on bottomside of the Watterott chip together. 
Any help would be nice

Chr157i4n, 2024-01-13:
Hello,

yes, the library should also work with other TMC2209 boards like yours.

Could you please share some more information?

Do you have the SilentStepStick TMC2209 v2?
https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC2209_v20.pdf

What is the output of the [demo/debug_script_01_uart_connection.py](https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/main/demo/debug_script_01_uart_connection.py) script?

Can you please share a sketch or photo of your wiring.

StBe121, 2024-01-14:
Thanks for the reply 
the output the script is 

---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: GPIO Init finished
TMC2209 0: Init finished
---
---
TMC2209 0: ---
TMC2209 0: TEST UART
TMC2209 0: received 12 bytes; 96 bits
TMC2209 0: hex: 550006e805ff0621000041c6
TMC2209 0: bin: 010101010000000000000110111010000000010111111111000001100010000100000000000000000100000111000110
TMC2209 0: length snd: 4
TMC2209 0: length rtn: 12
TMC2209 0: the Raspberry Pi received the sended
                            bits and the answer from the TMC
TMC2209 0: the Raspberry Pi received exactly the bits it has send.
                    the first 4 bits are the same
TMC2209 0: complete
TMC2209 0: 550006e8
TMC2209 0: 550006e805ff0621000041c6
TMC2209 0: just the first 4 bits
TMC2209 0: 550006e8
TMC2209 0: 550006e8
TMC2209 0: ---
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---

Yes i can confirm that that is the right stepper modul. 
I soldered the jumper pads together and checked the connection by resistence 
The only two requirements for the UART connection that I found was the solder connection to at least 2 of the pads(see picture) and an 1k resistor in TX wire 

Another qustion where can I set which UART channel the PI is using I would need to use 5-6 stepper and I want to use sensorless homing on all of them 
[image]
[image]
[image]
[image]

StBe121, 2024-01-14:
Ok made some more tests the output from the answer above is done without the external powersupply(PC power supply with 12/5/3.3V) as soon as its activ I get no connection  
th eoutput from your debuggfile is
---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: GPIO Init finished
TMC2209 0: Init finished
---
---
TMC2209 0: ---
TMC2209 0: TEST UART
TMC2209 0: received 12 bytes; 96 bits
TMC2209 0: hex: 550006e805ff0621000041c6
TMC2209 0: bin: 010101010000000000000110111010000000010111111111000001100010000100000000000000000100000111000110
TMC2209 0: length snd: 4
TMC2209 0: length rtn: 12
TMC2209 0: the Raspberry Pi received the sended
                            bits and the answer from the TMC
TMC2209 0: the Raspberry Pi received exactly the bits it has send.
                    the first 4 bits are the same
TMC2209 0: complete
TMC2209 0: 550006e8
TMC2209 0: 550006e805ff0621000041c6
TMC2209 0: just the first 4 bits
TMC2209 0: 550006e8
TMC2209 0: 550006e8
TMC2209 0: ---
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---

thats a little weird at least for me all devices share a common ground

Chr157i4n, 2024-01-14:
## Regarding your connection problem
I dont understand what exactly you mean with "cannot get a working connection", because in both cases your script output show the correct answer from the TMC (4 send + 8 received bytes = total 12 bytes) and the connection seems to work correctly.
```TMC2209 0: received 12 bytes; 96 bits```

what exactly does your power supply work in both cases? Normally you dont need the Vio Voltage, because the drivers should have internal voltage regulator, but i am not sure whether this is the case for every TMC2209 board. 

## Connecting 5-6 drivers via UART
connecting up to 5-6 drivers via UART to use StallGuard on each motor is tricky, because the addressing on one UART line support only 4 drivers:
[image]

As you can use the DIAG pin on the TMC as StallGuard Output, you could connect the DIAG of each driver to different pins of the RPi/ESP and use the "write only access" configuration:
[image]

If you want to have write and read access on each driver individually, you either need to use different UART port (for example SoftwareSerial on the ESP, on the RPi this is a bit harder and i have not tried that)
Or you use analog switches for the UART connection:
[image]

StBe121, 2024-01-14:
Thanks a lot for your time there were different fault yesterday I was able to recreate them 
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes
TMC2209: after 10 tries not valid answer. exiting
TMC2209: is Stepper Powersupply switched on ?
I think my old breadboard is worn out, I just touhced one wire and it happend again. 
As I said thanks for your support and for the lib it helps a lot also the documentations makes it easier to work with. Maybe I misinterpreted the print statements from today. Reread it again and yes I didnt understood them. 

I think the 5-6 steppers are for the ESP32 the RPI was only to test it in generell. Should be able to use a secound of the UARTs if the adress exceeds 3. the other lib from you and kjk25 uses at the moment hardcoded UART 2 right?

Chr157i4n, 2024-01-14:
Thank you very much. I try to improve the documentation with every feedback i get.

```TMC2209: did not get the expected 4 data bytes. Instead got 0 Bytes```
If you get 0 bytes, there is a problem with your wiring.
Because of the shared RX and TX line, the RPi/ESP should always at least receive the 4 bytes it has send to the TMC.

On the RPi you can set the SerialPort in the [constructor ](https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/12e549a00e2a7f16716d603618001a5f5bcdfe3d/src/TMC_2209/TMC_2209_StepperDriver.py#L104)with ```serialport="/dev/serial0"```

On the ESP it seems you can use ```serialport=2``` in the [constructor](https://github.com/kjk25/TMC2209_ESP32/blob/eb7fe3b1a6da3a923fd48ee8764366dc83e7dd21/tmc/TMC_2209_StepperDriver.py#L77) of kjk25 lib.

Chr157i4n, 2024-01-20:
I will close this issue now. If you still have questions feel free to reopen this issue or open a new one.

closed 2024-01-20 by Chr157i4n