Chr157i4n/PyTmcStepper #50: How to disable UART and use just STEP/DIR

iosifnicolae2, 2024-02-03:
**Describe the bug**
I would like to control [CL57T](https://www.omc-stepperonline.com/download/CL57T_V4.0.pdf) using this driver, but unfortunetly this stepper driver does not support uart.

Wouldn't it be possible to adapt your library to accept also stepper drivers without UART connection?

Thank you!

Chr157i4n, 2024-02-03:
Hey,

if you use the parameter ```skip_uart_init = True``` on the constructor and not use the functions that access the tmc registers, the library should also works with drivers that don't have UART.

Example:
```python
tmc = TMC_2209(21, 16, 20, skip_uart_init = True)

tmc.set_acceleration_fullstep(1000)
tmc.set_max_speed_fullstep(250)

tmc.set_motor_enabled(True)

tmc.run_to_position_steps(400)
tmc.run_to_position_steps(0)

tmc.set_motor_enabled(False)

del tmc
```

iosifnicolae2, 2024-02-03:
Hi,

Thank you for your quick response.

In the meantime, I've created a fork specific for CL5T:
https://github.com/iosifnicolae2/cl57t_raspberry_pi_stepper_driver

Hope that it's helpful for other people trying to work with this controller.

Chr157i4n, 2024-02-09:
I will close this issue now. If you still have questions feel free to reopen this issue or open a new one.

closed 2024-02-09 by Chr157i4n