Chr157i4n/PyTmcStepper #53: TWOTREES TMC2209 v2.0 passes tests but fails to energize stepper

leogolds, 2024-04-09:
**Describe the bug**
The basic tests seem to pass happily (pin, uart), but the stepper isn't moving when trying the basic movement demo. Doesn't even seem to be energized since I can turn it freely during movement.

Sanity checks performed:
* Measured resistance on both coils is 2 Ohm 
* Stepper spins freely without power
* Stepper Locks up when coil connected directly to the power supply
* Triple-checked wiring 
* Made sure each coil is connected to A/B terminals of 
* UART test seems to pass
* Pins test seems to pass

Any tips about how to proceed?
I tried the movement pin/uart debug and movement demo scripts on 3 different, brand-new drivers with the same result. 
Any other basic troubleshooting I might be missing?

```
python -m demo.debug_script_01_uart_connection

---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: EN Pin: 26
TMC2209 0: STEP Pin: 19
TMC2209 0: DIR Pin: 13
TMC2209 0: GPIO Init finished
TMC2209 0: Init finished
---
---
TMC2209 0: ---
TMC2209 0: TEST UART
TMC2209 0: received 12 bytes; 96 bits
TMC2209 0: hex: 550006e805ff0621000251a5
TMC2209 0: bin: 010101010000000000000110111010000000010111111111000001100010000100000000000000100101000110100101
TMC2209 0: length snd: 4
TMC2209 0: length rtn: 12
TMC2209 0: complete messages:
TMC2209 0: 550006e8
TMC2209 0: 550006e805ff0621000251a5
TMC2209 0: just the first 4 bits:
TMC2209 0: 550006e8
TMC2209 0: 550006e8
TMC2209 0: the Raspberry Pi received the sended
                            bits and the answer from the TMC
TMC2209 0: the Raspberry Pi received exactly the bits it has send.
                    the first 4 bits are the same
TMC2209 0: ---
TMC2209 0: UART connection: OK
TMC2209 0: ---
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---

python -m demo.demo_script_01_uart_connection

---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: GPIO Init finished
TMC2209 0: clearing GSTAT
TMC2209 0: Init finished
TMC2209 0: Init
TMC2209 0: EN Pin: 26
TMC2209 0: STEP Pin: 19
TMC2209 0: DIR Pin: 13
TMC2209 0: GPIO Init finished
TMC2209 0: clearing GSTAT
TMC2209 0: Init finished
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
TMC2209 0: write normal motor direction
TMC2209 0: activated 5V-out for current scale
TMC2209 0: CS too low; switching to VSense True
TMC2209 0: activated High sensitivity, low sense resistor voltage
TMC2209 0: cs_irun: 9
TMC2209 0: CS_IHold: 4
TMC2209 0: Delay: 10
TMC2209 0: actual current: 306 mA
TMC2209 0: ihold_irun: 0b10100000100100000100
TMC2209 0: writing ihold_irun
TMC2209 0: enabled PDN_UART
TMC2209 0: writing microstep interpolation setting: True
TMC2209 0: activated Stealthchop
TMC2209 0: writing 2 microstep setting
TMC2209 0: writing MStep Reg Select: True
TMC2209 0: activated operation with external sense resistors
---
---
TMC2209 0: ---
TMC2209 0: INPUTS
TMC2209 0: 0b100001000000000000001001010001
TMC2209 0: spread is low
TMC2209 0: dir is high
TMC2209 0: step is low
TMC2209 0: en is high
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: CHOPPER CONTROL
TMC2209 0: 0b10111000000110000000001010011
TMC2209 0: native 2 microstep setting
TMC2209 0: interpolation to 256 µsteps
TMC2209 0: 1: High sensitivity, low sense resistor voltage
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: DRIVER STATUS:
TMC2209 0: -0b111111111101110000000000000000
TMC2209 0: Info: motor is standing still
TMC2209 0: Info: motor is running on StealthChop
TMC2209 0: CS actual: 9
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: GENERAL CONFIG
TMC2209 0: 0b111000000
TMC2209 0: Driver is using internal reference derived from 5VOUT
TMC2209 0: Operation with external sense resistors
TMC2209 0: StealthChop PWM mode enabled
TMC2209 0: normal motor direction
TMC2209 0: INDEX shows the first microstep position of sequencer
TMC2209 0: INDEX output as selected by index_otpw
TMC2209 0: Microstep resolution selected by MSTEP register
TMC2209 0: ---
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---

python -m demo.demo_script_02_pin_connection

---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: GPIO Init finished
TMC2209 0: clearing GSTAT
TMC2209 0: Init finished
TMC2209 0: write normal motor direction
TMC2209 0: activated 5V-out for current scale
TMC2209 0: CS too low; switching to VSense True
TMC2209 0: activated High sensitivity, low sense resistor voltage
TMC2209 0: cs_irun: 9
TMC2209 0: CS_IHold: 4
TMC2209 0: Delay: 10
TMC2209 0: actual current: 306 mA
TMC2209 0: ihold_irun: 0b10100000100100000100
TMC2209 0: writing ihold_irun
TMC2209 0: enabled PDN_UART
TMC2209 0: writing microstep interpolation setting: True
TMC2209 0: activated Stealthchop
TMC2209 0: writing 2 microstep setting
TMC2209 0: writing MStep Reg Select: True
TMC2209 0: activated operation with external sense resistors
---
---
TMC2209 0: ---
TMC2209 0: INPUTS
TMC2209 0: 0b100001000000000000001011010001
TMC2209 0: spread is low
TMC2209 0: dir is high
TMC2209 0: step is high
TMC2209 0: en is high
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: INPUTS
TMC2209 0: 0b100001000000000000000001010000
TMC2209 0: spread is low
TMC2209 0: dir is low
TMC2209 0: step is low
TMC2209 0: en is low
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: INPUTS
TMC2209 0: 0b100001000000000000001011010001
TMC2209 0: spread is low
TMC2209 0: dir is high
TMC2209 0: step is high
TMC2209 0: en is high
TMC2209 0: ---
TMC2209 0: Motor output active: False
TMC2209 0: ---
TMC2209 0: Pin DIR:     OK
TMC2209 0: Pin STEP:    OK
TMC2209 0: Pin EN:      OK
TMC2209 0: ---
---
---
TMC2209 0: Motor output active: False
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---

python -m demo.demo_script_03_basic_movement

---
SCRIPT START
---
TMC2209 0: Init
TMC2209 0: EN Pin: 26
TMC2209 0: STEP Pin: 19
TMC2209 0: DIR Pin: 13
TMC2209 0: GPIO Init finished
TMC2209 0: clearing GSTAT
TMC2209 0: Init finished
TMC2209 0: write normal motor direction
TMC2209 0: activated 5V-out for current scale
TMC2209 0: CS too low; switching to VSense True
TMC2209 0: activated High sensitivity, low sense resistor voltage
TMC2209 0: cs_irun: 9
TMC2209 0: CS_IHold: 4
TMC2209 0: Delay: 10
TMC2209 0: actual current: 306 mA
TMC2209 0: ihold_irun: 0b10100000100100000100
TMC2209 0: writing ihold_irun
TMC2209 0: enabled PDN_UART
TMC2209 0: writing microstep interpolation setting: True
TMC2209 0: activated Stealthchop
TMC2209 0: writing 2 microstep setting
TMC2209 0: writing MStep Reg Select: True
TMC2209 0: activated operation with external sense resistors
---
---
TMC2209 0: ---
TMC2209 0: INPUTS
TMC2209 0: 0b100001000000000000001001010001
TMC2209 0: spread is low
TMC2209 0: dir is high
TMC2209 0: step is low
TMC2209 0: en is high
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: CHOPPER CONTROL
TMC2209 0: 0b10111000000110000000001010011
TMC2209 0: native 2 microstep setting
TMC2209 0: interpolation to 256 µsteps
TMC2209 0: 1: High sensitivity, low sense resistor voltage
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: DRIVER STATUS:
TMC2209 0: -0b111111111101110000000000000000
TMC2209 0: Info: motor is standing still
TMC2209 0: Info: motor is running on StealthChop
TMC2209 0: CS actual: 9
TMC2209 0: ---
TMC2209 0: ---
TMC2209 0: GENERAL CONFIG
TMC2209 0: 0b111000000
TMC2209 0: Driver is using internal reference derived from 5VOUT
TMC2209 0: Operation with external sense resistors
TMC2209 0: StealthChop PWM mode enabled
TMC2209 0: normal motor direction
TMC2209 0: INDEX shows the first microstep position of sequencer
TMC2209 0: INDEX output as selected by index_otpw
TMC2209 0: Microstep resolution selected by MSTEP register
TMC2209 0: ---
---
---
TMC2209 0: Motor output active: True
TMC2209 0: Motor output active: False
---
---
TMC2209 0: Deinit
TMC2209 0: Motor output active: False
TMC2209 0: GPIO cleanup
TMC2209 0: Deinit finished
---
SCRIPT FINISHED
---
```

**To Reproduce**
Steps to reproduce the behavior:
1. clone the repo
2. from repo root, run `python -m demo.demo_X.py

**Expected behavior**
Stepper should rotate, as per demo/demo_script_03_basic_movement.py

**Versions**
- Raspberry Pi 4
- Raspbian GNU/Linux 11 (bullseye)
- 0.4.3
- TWOTREES TMC2209 v2.0

**Additional context**
[image]
TMC2209 is powered using a variable power supply set to 12V
RPi powered seperately

Chr157i4n, 2024-04-09:
Hey, thanks for the detailed report.

- UART connection seems to work
- the STEP/DIR Interface seems to work aswell

So this should be an issue with your driver and not with the Scripts, Pi or Wiring.

Where did you buy your driver?
When i google "TWOTREES TMC2209 v2.0", i get a product from Aliexpress for under 1€, which say TNC2209 and Gerwany. So i guess that is a Scam.

I guess the driver does not output any voltage/current to the motor. On my driver there is an ENN pin (Motor current output gets activated with a LOW level on the pin), but on your driver there seems to be a EN Pin. Maybe the current output gets activated with a HIGH level instead.
You can test whether your driver activates by connecting the EN/ENN pin directly to GND or 3.3V.

leogolds, 2024-04-11:
Thank you so much for your response! I'll try that and report back.
In the meantime, I ordered some BTT-branded TMC2209 as a sanity check.

leogolds, 2024-04-17:
Got a batch of new BTT drivers which just worked. Once I had a sanity test, tried the TWOTREES again. 2 our of 3 drivers did not energize the stepper. The 3rd worked flawlessly when pulling ENN to Low. After running it for a bit, it stopped energizing the stepper. No such issues with the BTT's
Seems like I got a bad batch.

Thanks for the help @Chr157i4n, and the detailed information in this project!

closed 2024-04-17 by leogolds