Chr157i4n/PyTmcStepper #54: Request: Replace logger print-statement with python logging library

SiboVG, 2024-04-09:
Currently, logger message are just printed to the console. However, I believe it would be better to replace this with the `logging` library, as it offers more flexibility of where the log messages can be written to.

Chr157i4n, 2024-04-10:
I have to think about how i want to approach this. Either we switch from my own logger class to python-```logging``` completely. or we integrate python-```logging``` into the current logger class.
i don't have time this week, so it will probably take a little longer, or you could implement it and make a Pull request.

SiboVG, 2024-04-10:
> I have to think about how i want to approach this. Either we switch from my own logger class to python-`logging` completely. or we integrate python-`logging` into the current logger class. i don't [...]

Sure, I'll take a look at it. I'm also not sure which strategy would be best. It will probably become clear during development. Ideally, I'd want to maintain compatibility with previous versions so the function calls remain the same. We'll see, I'll keep you updated!

SiboVG, 2024-04-12:
Fixed by #55.

PR #55 (2024-04-12): [#54] Change logger to use logging module instead of print 
This PR fixes #54 by replacing the simple print messages in the TMC logger by the python `logging` library. The original TMC logging class remains unchanged, as do its pre-existing function calls, so this update is backward-compatible with code from previous versions. The only thing I had to change was the Loglevel values to make them comply to the default values from `logging`. But I don't image that was actually used other than internally.

The`TMC_logger` now stores a `logging` instance to which all its log messages are saved. The  [... cut]
closed 2024-04-12 by SiboVG