Chr157i4n/PyTmcStepper #59: RuntimeError when redoing homing after failed attempt

SiboVG, 2024-04-16:
**Describe the bug**
I am running the `do_homing` method, but when I try to rerun the homing after a failed attempt, I get this error:
```Python
╭─────────────────────────────────────────────────────────────────────── Traceback (most recent call last) ────────────────────────────────────────────────────────────────────────╮
│ /home/pi/Downloads/dip_coater/src/dip_coater/tui.py:267 in do_homing_action                                                                                                      │
│                                                                                                                                                                                  │
│    264 │   │   │   log.write(f"[cyan]Doing homing ({revs=}, {threshold=}, {speed=} RPM)...[/]") ╭───────────── locals ─────────────╮                                             │
│    265 │   │   │   self.set_motor_state("homing")                                               │       log = RichLog(id='logger') │                                             │
│    266 │   │   │   await asyncio.sleep(0.1)                                                     │      revs = 10                   │                                             │
│ ❱  267 │   │   │   self.motor_driver.do_homing(revs, threshold, speed)                          │      self = MotorControls()      │                                             │
│    268 │   │   │   log.write("-> Finished homing.")                                             │     speed = 75                   │                                             │
│    269 │   │   │   self.set_motor_state("enabled")                                              │ threshold = 190                  │                                             │
│    270 │   │   elif self._motor_state == "homing":                                              ╰──────────────────────────────────╯                                             │
│                                                                                                                                                                                  │
│ /home/pi/Downloads/dip_coater/src/dip_coater/motor.py:145 in do_homing                                                                                                           │
│                                                                                                                                                                                  │
│   142 │   │   """                                                                              ╭────────────────────────────────── locals ──────────────────────────────────╮    │
│   143 │   │   # Homing sets the spreadcycle to StealthChop, so we need to store the original s │  revolutions = 10                                                          │    │
│   144 │   │   spread_cycle = self.tmc.get_spreadcycle()                                        │         self = <dip_coater.motor.TMC2209_MotorDriver object at 0xb5553100> │    │
│ ❱ 145 │   │   self.tmc.do_homing(                                                              │    speed_rpm = 75                                                          │    │
│   146 │   │   │   diag_pin=self.diag_pin,                                                      │ spread_cycle = 0                                                           │    │
│   147 │   │   │   revolutions=revolutions,                                                     │    threshold = 190                                                         │    │
│   148 │   │   │   threshold=threshold,                                                         ╰────────────────────────────────────────────────────────────────────────────╯    │
│                                                                                                                                                                                  │
│ /home/pi/.local/lib/python3.8/site-packages/TMC_2209/TMC_2209_StepperDriver.py:230 in do_homing                                                                                  │
│                                                                                                                                                                                  │
│   227 │   │                                                                                    ╭─────────────────────────────────── locals ────────────────────────────────────╮ │
│   228 │   │   self.tmc_logger.log(f"Stallguard threshold: {self._sg_threshold}", Loglevel.DEBU │    diag_pin = 26                                                              │ │
│   229 │   │                                                                                    │ revolutions = 10                                                              │ │
│ ❱ 230 │   │   self.set_stallguard_callback(diag_pin, self._sg_threshold, self.stop,            │        self = <TMC_2209.TMC_2209_StepperDriver.TMC_2209 object at 0xb5553148> │ │
│   231 │   │   │   │   │   │   │   │   │    0.5*tmc_math.rps_to_steps(speed_rpm/60, self._steps │   speed_rpm = 75                                                              │ │
│   232 │   │                                                                                    │   threshold = 190                                                             │ │
│   233 │   │   homing_failed = self.set_vactual_rpm(speed_rpm, revolutions=revolutions)         ╰───────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                                                                                                                  │
│ /home/pi/.local/lib/python3.8/site-packages/TMC_2209/TMC_2209_StepperDriver.py:486 in set_stallguard_callback                                                                    │
│                                                                                                                                                                                  │
│   483 │   │                                                                                                                                                                      │
│   484 │   │   GPIO.setup(self._pin_stallguard, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)                                                                                              │
│   485 │   │                                                                                                                                                                      │
│ ❱ 486 │   │   GPIO.add_event_detect(self._pin_stallguard, GPIO.RISING, callback=self.stallguar                                                                                   │
│   487 │   │   │   │   │   │   │     bouncetime=300)                                                                                                                              │
│   488                                                                                                                                                                            │
│   489                                                                                                                                                                            │
│                                                                                                                                                                                  │
│ ╭──────────────────────────────────────────────── locals ─────────────────────────────────────────────────╮                                                                      │
│ │       callback = <bound method stop of <TMC_2209.TMC_2209_StepperDriver.TMC_2209 object at 0xb5553148>> │                                                                      │
│ │      min_speed = 1000.0                                                                                 │                                                                      │
│ │ pin_stallguard = 26                                                                                     │                                                                      │
│ │           self = <TMC_2209.TMC_2209_StepperDriver.TMC_2209 object at 0xb5553148>                        │                                                                      │
[... cut ...]
RuntimeError: Conflicting edge detection already enabled for this GPIO channel
```

**To Reproduce**
Steps to reproduce the behavior:
1. installed the library with pip
2. ran the function 'do_homing'
4. Homing failed because I did not set the revolutions parameter high enough
5. Retried homing
6. Got the error

**Expected behavior**
The event should be removed from the diag pin after homing has finished.

**Versions**
- Raspberry Pi Version 4 model B
- OS Version
- Version of this library 0.4.4
- TMC2209 1.3 Bigtreetech

Chr157i4n, 2024-04-16:
The event should get removed in the destructor:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/b3aaf7d76c52dd59a61167c83789ee89b90818fc/src/TMC_2209/TMC_2209_StepperDriver.py#L173-L175

I will try to reproduce it.

SiboVG, 2024-04-16:
> The event should get removed in the destructor:
> 
> https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/b3aaf7d76c52dd59a61167c83789ee89b90818fc/src/TMC_2209/TMC_2209_StepperDriver.py#L173-L175
> 
> I will try to reproduce it.

But that's not exactly desirable, right? That means that instead of calling `do_homing` again on my tmc instance (because it failed, I want to rerun it until success), I have to first delete the tmc instance and then create a new tmc instance on which I call `do_homing` again?

Chr157i4n, 2024-04-16:
If i understand your problem correctly, you want to execute do_homing twice in a row and this is not possible because in the do_homing function the event_detect is added, but not removed?

SiboVG, 2024-04-16:
> If i understand your problem correctly, you want to execute do_homing twice in a row and this is not possible because in the do_homing function the event_detect is added, but not removed?

That's right.

SiboVG, 2024-04-16:
Either the event_detect should be removed, or the program should check whether the event is already present when attempting to add it.

commit f0c66c74 (2024-04-16): remove existing RPi.GPIO event_detect before adding a new one
- fixes #59