Chr157i4n/PyTmcStepper #64: Add support for Raspberry Pi 5

SiboVG, 2024-05-06:
Today I wanted to test out my TMC2209 driver code on a RPi 5, only to find out that `RPi.GPIO` does not support the new RPi GPIO hardware.

I would therefore propose to either replace `RPi.GPIO` entirely with the `gpiozero` library, or to provide some sort of flag that can switch between `RPi.GPIO` (since a lot of users still use this library) and `gpiozero` (for RPi 5 users). I guess the latter is more flexible for users, but more tricky to implement.

Reason for choosing `gpiozero`: it is more high-level than for instance `gpiod`, but still supports pretty much all the functions that RPi.GPIO has.

Chr157i4n, 2024-05-06:
Seems like RPi.GPIO will not be updated for the RPi 5 in the near future:
https://sourceforge.net/p/raspberry-gpio-python/tickets/214/

1. That will be a problem with Nvidia Jetson support:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/pull/47
2. Both [gpiozero](https://github.com/gpiozero/gpiozero) and [gpiod](https://github.com/hhk7734/python3-gpiod) are written in pure Python and should be therefore slower than RPi.GPIO.

I will check at the Weekend how good the libs works on my Pi4. If the speed is a problem, we can implement a "Loose coupling" and still use RPi.GPIO for the Pi4 and a different lib for the Pi5.

SiboVG, 2024-05-06:
> 2\. Both [gpiozero](https://github.com/gpiozero/gpiozero) and [gpiod](https://github.com/hhk7734/python3-gpiod) are written in pure Python and should be therefore slower than RPi.GPIO.

Hm crap, was not aware of that... `libgpiod` (writtin in C) does provide Python bindings, instead of being written in pure Python. Here is an interesting speed comparison of `gpiozero` and `libgpiod`: [link](https://adafruit-playground.com/u/MakerMelissa/pages/comparing-libgpiod-and-gpiozero-speeds-on-the-raspberry-pi-5). `libgpiod` is 3x faster.

Chr157i4n, 2024-05-06:
I am pretty sure that the calculation, of the time interval until the next step has to occur (also written in pure Python) , has more impact on the maximum possible motor speed then the actual gpio interaction.
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/9e14313f59c4f45ecef1d77665044d40cf6427e5/src/TMC_2209/_TMC_2209_move.py#L301C1-L371

If i remember correctly, getting the current time has the most impact:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/9e14313f59c4f45ecef1d77665044d40cf6427e5/src/TMC_2209/_TMC_2209_move.py#L381

SiboVG, 2024-05-06:
I guess something like [RPi.GPIO2](https://github.com/underground-software/RPi.GPIO2) would be ideal, but it hasn't been maintained for some time, and there appear to be some major bugs in the library.

SiboVG, 2024-05-06:
> I am pretty sure that the calculation, of the time interval until the next step has to occur (also written in pure Python) , has more impact on the maximum possible motor speed then the actual gpio [...]

Would [time.perf_counter()](https://docs.python.org/3.5/library/time.html#time.perf_counter) be faster?

Chr157i4n, 2024-05-18:
I moved the GPIO lib access into the _TMC_2209_GPIO_board file and the TMC_gpio class. So we can implemented different library easily.
Can you please confirm that the file `/proc/device-tree/model` can be used to identify a Raspberry Pi 5.

For me (on RPi4) this file contains the exact model name
```
cat /proc/device-tree/model

Raspberry Pi 4 Model B Rev 1.2
```

Chr157i4n, 2024-05-18:
I implemented a basic version for the `demo_script_03_basic_movement` script of gpiozero for the RPi5. Can you test that code for me?
Maybe i should buy one myself.

SiboVG, 2024-05-18:
> I implemented a basic version for the `demo_script_03_basic_movement` script of gpiozero for the RPi5. Can you test that code for me? Maybe i should buy one myself.

I can maybe test it next week. Again, I do think that we should go with `gpiod` (not the pure Python version, but the Python-bindings of the original C lib - on [PyPI](https://pypi.org/project/gpiod/) this is starting from version 2.0.2)

Chr157i4n, 2024-05-18:
For me libgpio is even slower than gpiozero with 25 sec for the 1.000.000 toggles.

```
RPi.GPIO
1.4 sec

libgpio
25 sec

gpiozero
18 sec
```

I attached the 3 scripts for the benchmark of the three libs:
[python_gpio_lib_benchmark.zip](https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/files/15366108/python_gpio_lib_benchmark.zip)

i give up trying to get RPi.GPIO2 to work. It has a long list of requirements and there are a lot of "ERROR: Could not find a version that satisfies" Error on my RPi4 with Debian 11 Bullseye and Python 3.9.2.
As far as i understand RPi.GPIO is just RPi.GPIO syntax, but uses libgpiod in background. If libgpiod would be nearly as fast as the original RPi.GPIO this would be nice. But with the result of my benchmark, i would rather use gpiozero, which is preinstalled on the Raspberry Pi OS.

But please test the libs on your own next week and give me your feedback

SiboVG, 2024-05-19:
I was able to try it out today. Had to make some small modifications to the code (I pushed them to the `dev` branch), most importantly that the serial port for RPi 5 is `/dev/ttyAMA0` instead of `/dev/serial0`.

Benchmark results:
```
gpiozero
8 sec

gpiod
5 sec

(I can't test RPi.GPIO on an RPi 5)
```

So it appears that `gpiod` is optimized for the RPi 5 and is faster than `gpiozero`, contrary to your test results.

I will see if I can implement RPi.GPIO2 and see how that compares to the others.

> As far as i understand RPi.GPIO is just RPi.GPIO syntax, but uses libgpiod in background.

That's right.

SiboVG, 2024-05-20:
Also tested [rpi-lgpio](https://github.com/waveform80/rpi-lgpio?tab=readme-ov-file), which basically does the same thing as `RPi.GPIO2`:
```
rpi-lgpio
8 sec
```

`RPi.GPIO2` also gave a lot of installation warnings for me, so best not to use this.

Ideally, we would use `gpiozero` because it's natively installed, with a pin factory for `libgpiod` (see [this issue](https://github.com/gpiozero/gpiozero/issues/840)), but that is not yet supported...

jenswes, 2024-05-20:
Hello,
whats about https://github.com/vsergeev/python-periphery
i'm using a Luckfox Pico Max https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO
a nice device for me, as i want Ethernet an onboard Flash. i've installed alpine linux and they use periphery for GPIO

```
from periphery import GPIO
import time
from timeit import default_timer as timer
Write_Pin = 55  
Write_GPIO = GPIO(Write_Pin, "out")
start = timer()

for x in range(1000000):
    Write_GPIO.write(True)
    Write_GPIO.write(False)

end = timer()
print(end - start)
```
------------------------------------
jalpico:~/src# python test_periphery.py 
14.812718628001676

as this is a single core Processor 
Cortex A7 1.2GHz (Rockchip RV1106) 
256MB DDR3L
SPI NAND FLASH (256MB)

~15sec sounds not bad
but periphery is also written in pure Python

it would be nice to be able to use [TMC2209_Raspberry_Pi] on other platforms too

SiboVG, 2024-05-20:
> Luckfox Pico Max

I guess this should be a separate issue from this one. This is for supporting an RPi 5. Can you make a new issue for supporting the Luckfox Pico Max board @jenswes?

jenswes, 2024-05-20:
done :-)
i' just popping into this issue, as a new way supporting the Pi 5 is searched

RPi.GPIO
1.4 sec

libgpio
25 sec

gpiozero
18 sec

here python-periphery wth 15 sec on my small slow device looks not bad for the Pi too

but you are right, my issue ist support for Luckfox Pico (Max)

SiboVG, 2024-05-20:
> done :-)

Thanks!

Chr157i4n, 2024-05-25:
GPIO access and Egde detect for StallGuard is now implemented for the Pi5. If there is still something missing, let me know.
I will propably push this to main later today (and make a new release).

closed 2024-05-26 by Chr157i4n