Chr157i4n/PyTmcStepper #66: Support for Luckfox Pico maybe thru python-periphery support

jenswes, 2024-05-20:
Hello,
whats about https://github.com/vsergeev/python-periphery
i'm using a Luckfox Pico Max https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO
a nice device for me, as i want Ethernet an onboard Flash. i've installed alpine linux and they use periphery for GPIO

```
from periphery import GPIO
import time
from timeit import default_timer as timer
Write_Pin = 55  
Write_GPIO = GPIO(Write_Pin, "out")
start = timer()

for x in range(1000000):
    Write_GPIO.write(True)
    Write_GPIO.write(False)

end = timer()
print(end - start)
jalpico:~/src# python test_periphery.py
14.812718628001676

```
as this is a single core Processor
Cortex A7 1.2GHz (Rockchip RV1106)
256MB DDR3L
SPI NAND FLASH (256MB)

~15sec sounds not bad
periphery is also written in pure Python

it would be nice to be able to use [TMC2209_Raspberry_Pi] on other platforms too

thanks
jens

Chr157i4n, 2024-05-20:
Hey,

that should be possible to implement.
Could you please give us the output of
`cat /proc/device-tree/model`
to detect your board programatically.

jenswes, 2024-05-20:
sure :-)
cat /proc/device-tree/model
Luckfox Pico Max

i would suggest detecting only Luckfox Pico, all versions should fit.

additional i see here 
/dev/gpiochip0  
/dev/gpiochip1  
/dev/gpiochip2  
/dev/gpiochip3  
/dev/gpiochip4

but 26 pins should be exposed

also i've tried gpiod:
```
import gpiod
from timeit import default_timer as timer
test_pin = 9

pin_line =  gpiod.request_lines("/dev/gpiochip0",
                                consumer="blink-example",
                                config={
                                    test_pin: gpiod.LineSettings(
                                        direction=gpiod.line.Direction.OUTPUT,
                                        output_value=gpiod.line.Value.ACTIVE
                                        )
                                        }

start = timer()

pin_line.set_value(test_pin, gpiod.line.Value.INACTIVE)

for x in range(1000000):
    pin_line.set_value(test_pin, gpiod.line.Value.ACTIVE)
    pin_line.set_value(test_pin, gpiod.line.Value.INACTIVE)

end = timer()
```
---------------------------------------
 python test_libgpio_j0-test.py
25.914991062000013

and writing to sysfs with  https://github.com/vitiral/gpio
```
import time
from timeit import default_timer as timer
import gpio as GPIO

GPIO.setup(55, GPIO.OUT)

start = timer()

for x in range(1000000):
     GPIO.output(55, GPIO.HIGH)
     GPIO.output(55, GPIO.LOW)

end = timer()
print(end - start)
```
---------------------------------
jalpico:~/src# echo 55 > /sys/class/gpio/export ## has to be done first or you get Permission denied: '/sys/class/gpio/export'
jalpico:~/src# python test_sysfs-gpio.py
15.870769970000083

also not bad.

by looking into the dts file from kernel, i see a lot Hardware PWM (12?)  4 of them are enabled by default. Maybe this is an option instead of creating square wave with gpio bit-bang but thats an other issue as well :-)

Chr157i4n, 2024-05-21:
1. could you test the luckfox pico support on the dev branch?
2. PWM does not really work if you want to control the motor speed. You change the duty cycle of the PWM, but a change in the frequency of the pulses is required to control the motor speed

jenswes, 2024-05-22:
Hi, 
thanks :-)

Installing was a challange, it complains on missing RPi.GPIO
```

Looking in links: ./dist/
Processing ./dist/TMC_2209_Raspberry_Pi-0.4.5-py3-none-any.whl
INFO: pip is looking at multiple versions of tmc-2209-raspberry-pi to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement RPi.GPIO (from tmc-2209-raspberry-pi) (from versions: none)
ERROR: No matching distribution found for RPi.GPIO

```

so i use
`
pip3 install -v --break-system-packages --no-deps --no-index --find-links=./dist/ TMC_2209_Raspberry_Pi
`

From https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO
```
#
# As an example, let's consider GPIO1_C7_d, where:
#   bank : 1
#   group : 2 (A=0, B=1, C=2, D=3)
#   X : 7
#    So, the pin number of GPIO1_C7_d is: 1 x 32 + (2 x 8 + 7) = 55
#
pin 14 GPIO1 C1_d   1 x 32 + (2 x 8 + 1)  49   Step
pin 15 GPIO1 C2_d   1 x 32 + (2 x 8 + 2)  50   Dir
pin 16 GPIO1 C3_d   1 x 32 + (2 x 8 + 3)  51   Enable

 tmc = TMC_2209(51, 49, 50, serialport="/dev/ttyS3")  
```
but My GPIO Pin number are too high:
```
  File "/usr/lib/python3.11/site-packages/TMC_2209/_TMC_2209_GPIO_board.py", line 226, in gpio_output
    TMC_gpio._gpios[pin].write(bool(value))
    ~~~~~~~~~~~~~~~^^^^^
IndexError: list index out of range

```
 so i entered lower (wrong) ones

`      tmc = TMC_2209(19, 17, 18, serialport="/dev/ttyS3")`

```

---
SCRIPT START
---
2024-05-22 21:43:10,264 - TMC2209 0 - INFO - Init
2024-05-22 21:43:10,283 - TMC2209 0 - INFO - GPIO Init finished
2024-05-22 21:43:10,470 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    4 total bytes
```
console shows
```
of_dma_request_slave_channel: dma-names property of node '/serial@ff4d0000' missing or empty
dw-apb-uart ff4d0000.serial: failed to request DMA, use interrupt mode

```
but interrupt should be OK also :-)

if i connect a Terminal to /dev/ttyS3 and run demo_script_01_uart_connection.py
i see 
`UlMUlMUlMUlMUlMUlMUlMUlMUlMUlMUFUFUFUFUFUFUFUFUFUF`
(linux GTKterm) so the serial seems to work

I'm using some TMC2209 MKS V1.0, i'll try them next on a Pi4 to see if that works
or getting the bigtreetech ones.

My jumpercables are a bit lose, but when i measue all seems OK the 1k resistor as well :-)

thanks again for your work.

Chr157i4n, 2024-05-23:
1. I removed RPi.GPIO as required dependency.
Now the lib can be installled (as soon as it is on PyPi) with
`pip3 install TMC-2209-Raspberry-Pi[RASPBERRY_PI]` to get RPi.GPIO installed or
`pip3 install TMC-2209-Raspberry-Pi[LUCKFOX_PICO]` to get python-periphery installed with the lib.
If the lib is installed without any GPIO lib, it will give the according error on runtime.
2. I increased the number of possible pins to 200, so you can use your pin 55. I dont really understand why the luckfox uses that weird pin numbering and not just 1 to 40.

jenswes, 2024-05-25:
Hello, thanks a lot :-)
all is working fine.
RTFM is always good, if things not working
better not trying to use GPIOS that are redefined as SPI by the dts :-)

I've modded a 28byj-48 to bipolar according here:
https://ardufocus.com/howto/28byj-48-bipolar-hw-mod/

this motor has a gearbox, so movement is quite slow.

All sample move nicely.

commit 5e54ce3d (2024-05-21): initial support for the luckfox pico
resolves #66
closed 2024-05-26 by Chr157i4n