Chr157i4n/PyTmcStepper #68: DIR Pin not OK

Arturex0, 2024-06-11:
After many tries, the DIR pin refuses to work. I am certain that this is not a wiring issue, as it works with like 2 drivers that I have, but not with 6 other. How to troubleshoot?  

---
SCRIPT START
---
2024-06-10 17:37:49,133 - TMC2209 2 - INFO - Init
2024-06-10 17:37:49,133 - TMC2209 2 - INFO - GPIO Init finished
2024-06-10 17:37:49,143 - TMC2209 2 - INFO - clearing GSTAT
2024-06-10 17:37:49,175 - TMC2209 2 - INFO - Init finished
2024-06-10 17:37:49,184 - TMC2209 2 - INFO - write normal motor direction
2024-06-10 17:37:49,216 - TMC2209 2 - INFO - activated 5V-out for current scale
2024-06-10 17:37:49,238 - TMC2209 2 - INFO - CS too low; switching to VSense True
2024-06-10 17:37:49,247 - TMC2209 2 - INFO - activated High sensitivity, low sense resistor voltage
2024-06-10 17:37:49,270 - TMC2209 2 - INFO - cs_irun: 9
2024-06-10 17:37:49,270 - TMC2209 2 - INFO - CS_IHold: 4
2024-06-10 17:37:49,271 - TMC2209 2 - INFO - Delay: 10
2024-06-10 17:37:49,271 - TMC2209 2 - INFO - actual current: 306 mA
2024-06-10 17:37:49,271 - TMC2209 2 - INFO - ihold_irun: 0b10100000100100000100
2024-06-10 17:37:49,271 - TMC2209 2 - INFO - writing ihold_irun
2024-06-10 17:37:49,302 - TMC2209 2 - INFO - enabled PDN_UART
2024-06-10 17:37:49,334 - TMC2209 2 - INFO - writing microstep interpolation setting: True
2024-06-10 17:37:49,366 - TMC2209 2 - INFO - activated Stealthchop
2024-06-10 17:37:49,397 - TMC2209 2 - INFO - writing 2 microstep setting
2024-06-10 17:37:49,429 - TMC2209 2 - INFO - writing MStep Reg Select: True
2024-06-10 17:37:49,461 - TMC2209 2 - INFO - activated operation with external sense resistors
---
---
2024-06-10 17:37:49,584 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,584 - TMC2209 2 - INFO - INPUTS
2024-06-10 17:37:49,593 - TMC2209 2 - INFO - 0b100001000000000000000011001001
2024-06-10 17:37:49,593 - TMC2209 2 - INFO - spread is low
2024-06-10 17:37:49,593 - TMC2209 2 - INFO - dir is low
2024-06-10 17:37:49,594 - TMC2209 2 - INFO - step is high
2024-06-10 17:37:49,594 - TMC2209 2 - INFO - en is high
2024-06-10 17:37:49,594 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,694 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,694 - TMC2209 2 - INFO - INPUTS
2024-06-10 17:37:49,703 - TMC2209 2 - INFO - 0b100001000000000000000001001000
2024-06-10 17:37:49,703 - TMC2209 2 - INFO - spread is low
2024-06-10 17:37:49,704 - TMC2209 2 - INFO - dir is low
2024-06-10 17:37:49,704 - TMC2209 2 - INFO - step is low
2024-06-10 17:37:49,704 - TMC2209 2 - INFO - en is low
2024-06-10 17:37:49,704 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,804 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,804 - TMC2209 2 - INFO - INPUTS
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - 0b100001000000000000000011001001
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - spread is low
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - dir is low
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - step is high
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - en is high
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - Motor output active: False
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - ---
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - Pin DIR:   not OK
2024-06-10 17:37:49,814 - TMC2209 2 - INFO - Pin STEP:  OK
2024-06-10 17:37:49,815 - TMC2209 2 - INFO - Pin EN:    OK
2024-06-10 17:37:49,815 - TMC2209 2 - INFO - ---
---
---
2024-06-10 17:37:49,815 - TMC2209 2 - INFO - Motor output active: False
---
---
2024-06-10 17:37:49,815 - TMC2209 2 - INFO - Deinit
2024-06-10 17:37:49,819 - TMC2209 2 - INFO - Motor output active: False
2024-06-10 17:37:49,819 - TMC2209 2 - INFO - GPIO cleanup
2024-06-10 17:37:49,819 - TMC2209 2 - INFO - Deinit finished
---
SCRIPT FINISHED
---

SiboVG, 2024-06-11:
Have you verified that you can pull the GPIO pin high or low:

```Python
GPIO.setup(20, GPIO.OUT)
GPIO.output(20, GPIO.HIGH)
```

I had the same issue a while ago where the GPIO pin on my Raspberry Pi wasn't working.

I also recommend that you try and change the GPIO number of the DIR pin to something other than 20, and see whether that works.

Chr157i4n, 2024-06-25:
Does this issue still persists? Otherwise this issue will be closed at the end of this week.

closed 2024-06-30 by Chr157i4n