Chr157i4n/PyTmcStepper #69: Can't turn UART connection off

sin-2pi, 2024-06-24:
Using a Raspberry Pi 5. Latest Raspberry Pi OS. Even when putting the 'skip_uart_init = True' flag into the tmc constructor, this is still coming up. I even tried connecting the TX/RX pins yet it is still giving this error. I just want to connect via pins directly, no UART. I am running the demo_script_02_pin_connection.py
`---
SCRIPT START
---
2024-06-24 15:51:54,063 - TMC2209 0 - INFO - Init
2024-06-24 15:51:54,069 - TMC2209 0 - INFO - GPIO Init finished
2024-06-24 15:51:54,069 - TMC2209 0 - INFO - Init finished
2024-06-24 15:51:54,252 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    1 total bytes
2024-06-24 15:51:54,435 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:54,618 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:54,800 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:54,983 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,166 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,349 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,532 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,715 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,898 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:55,898 - TMC2209 0 - ERROR - after 10 tries not valid answer
2024-06-24 15:51:56,081 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:56,263 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:56,446 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:56,630 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:56,813 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:56,996 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:57,179 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:57,363 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:57,545 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:57,728 - TMC2209 0 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    0 total bytes
2024-06-24 15:51:57,728 - TMC2209 0 - ERROR - after 10 tries not valid answer
2024-06-24 15:51:57,728 - TMC2209 0 - INFO - EXITING!
2024-06-24 15:51:57,841 - TMC2209 0 - INFO - Deinit
Exception ignored in: <function TMC_2209.__del__ at 0x7fffb558dee0>`

Chr157i4n, 2024-06-25:
the `skip_uart_init` flag only lets you skip the UART communication in the init function. Since you have `TMC2209 0 - INFO - Init finished` in the output, this also works for you.
But there are other commands in the demo script that are using UART. The whole script `demo_script_02_pin_connection.py` pulls the pins to high/low and checks via UART whether the TMC recognizes them as high/low.

In order to get the `demo_script_02_pin_connection.py` for you to work, you need to comment out the following lines (the basically does nothing then ;) ):
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/734c93e756e953f71883773e00873c35c217686c/demo/demo_script_02_pin_connection.py#L62-L87

for the other demo script [demo_script_03_basic_movement.py](https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/main/demo/demo_script_03_basic_movement.py) you need to comment out the following lines:
https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/blob/734c93e756e953f71883773e00873c35c217686c/demo/demo_script_03_basic_movement.py#L60-L87

sin-2pi, 2024-06-25:
Thanks for the response. It ended up that it was just a wiring issue. Everything works as it should now!

closed 2024-06-25 by sin-2pi