Chr157i4n/PyTmcStepper #82: Multiple Drivers: UART Communication Error

stahlbauer, 2024-12-14:
## Context And Observed Behavior

I have connected two TMC2209 (BigTreeTech) to a Raspberry Pi5. Both are connected to the same UART pins (TX/RX), with one 1k resistor, and the step/dir/enable pins. 

The motor stepping only works if only one of the drivers is initialized in the code. For example, the following two alternatives work:

```
tmc = TMC_2209(21, 16, 20, serialport="/dev/ttyAMA0", driver_address=2, loglevel=Loglevel.DEBUG)
```
and 

```
tmc2 = TMC_2209(13, 5, 6, serialport="/dev/ttyAMA0", driver_address=1, loglevel=Loglevel.DEBUG)
```
However, as soon as I enable both (and creating the additional `tmc2` object is sufficient here), I run into UART problems, that is
```
tmc = TMC_2209(21, 16, 20, serialport="/dev/ttyAMA0", driver_address=2, loglevel=Loglevel.DEBUG)
tmc2 = TMC_2209(13, 5, 6, serialport="/dev/ttyAMA0", driver_address=1, loglevel=Loglevel.DEBUG)
```
yields
```
python3 -m demo.demo_script_03_basic_movement.py
---
SCRIPT START
---
2024-12-14 16:40:28,465 - TMC2209 2 - INFO - Init
2024-12-14 16:40:28,465 - TMC2209 2 - DEBUG - EN Pin: 21
2024-12-14 16:40:28,469 - TMC2209 2 - DEBUG - STEP Pin: 16
2024-12-14 16:40:28,469 - TMC2209 2 - DEBUG - DIR Pin: 20
2024-12-14 16:40:28,469 - TMC2209 2 - INFO - GPIO Init finished
2024-12-14 16:40:28,652 - TMC2209 2 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    4 total bytes
2024-12-14 16:40:28,835 - TMC2209 2 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    4 total bytes
2024-12-14 16:40:29,017 - TMC2209 2 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    4 total bytes
2024-12-14 16:40:29,200 - TMC2209 2 - ERROR - UART Communication Error:
                                    0 data bytes |
                                    4 total bytes
2024-12-14 16:40:29,383 - TMC2209 2 - ERROR - UART Communication Error:
```
An interesting observation here is, that only the driver at address `2` reports errors; might be, because `tmc2` (with address `1`) is actually not used.

The script `demo_script_06_multiple_drivers.py` indicates that `Multiple driver not tested`. Is this still the case? 

## Steps To Reproduce

Adjust `demo_script_03_basic_movement` and add `tmc2` as the second driver object, and have a corresponding driver connected. 

## Expected Behavior

The library works with several several drivers.

## Versions

- Raspberry Pi Version 5
- Version of this library: Oct 6th
- TMC2209 version and manufacturer: BigTreeTech 1.3

Chr157i4n, 2024-12-14:
Hey,

did you set the uart address of the tmc with the MS1_AD0 and MS2_AD1 pins?
Please have a look at https://github.com/Chr157i4n/TMC2209_Raspberry_Pi/wiki/Multiple-Drivers

If you set it up like this:
[image]
you need to use the driver_address 0 and 1 and not 1 and 2 in the library.

stahlbauer, 2024-12-15:
Hey. Thanks a lot for your reply. I did set the addresses correctly. However, I did have a voltage mismatch. Some logic pin were at 5 and some at 3.3V. Getting this right (for now, all at 3.3V) fixed the issue and I can address multiple drivers properly. 

Thanks again. Your library is great (even if I would prefer a less restrictive license).

closed 2024-12-15 by stahlbauer