Chr157i4n/PyTmcStepper #103: TMC2209 “run_to_position_fullsteps(ABSOLUTE)” Not Moving to Correct Position

dcl920108, 2025-03-20:
### **Title:**  
TMC2209 `run_to_position_fullsteps(ABSOLUTE)` Not Moving to Correct Position  

---

### **Describe the bug**  
When using `run_to_position_fullsteps(target, MovementAbsRel.ABSOLUTE)`, the stepper motor does not move to the expected absolute position. Instead, it sometimes reverses direction unexpectedly, or remains in place.  

---

### **To Reproduce**  
Steps to reproduce the behavior:  
1. Installed the library using `pip install PyTmcStepper`  
2. Ran the following test script:  
   ```python
   tmc.run_to_position_fullsteps(-2000, MovementAbsRel.ABSOLUTE)
   tmc.run_to_position_fullsteps(0, MovementAbsRel.ABSOLUTE)
   ```
3. Observed that the motor either:  
   - Moves but reverses unexpectedly  
   - Stays at its previous position and does not return to 0  
   - Runs inconsistently on repeated executions  

---

### **Expected behavior**  
- The motor should move directly to `-2000` steps and then return to `0` without reversing direction unexpectedly.  
- The step count should be maintained correctly across function calls.  

---

### **Versions**  
- **Board model:** Raspberry Pi 4 Model B  
- **Board model version:** Rev 1.5  
- **OS Version:** Raspberry Pi OS (Debian-based)  
- **Library version:** `PyTmcStepper` (latest pip release)  
- **TMC2209 version:** Generic TMC2209 (BIGTREETECH TMC2209 V1.3)  

---

### **Additional context**  
- Using `MovementAbsRel.RELATIVE` results in accumulated movement but also reverses direction unpredictably.  
- Manually setting `tmc.tmc_mc.current_pos = -2000` before the second move does not seem to fix the issue.  
- Tried different `microstepping` and `spreadcycle` settings, but the issue persists.  

Would appreciate any guidance on whether this is a library issue or if additional steps are needed to reset motor position correctly.

Chr157i4n, 2025-03-20:
which MotionControl class are you using?
I just tested your test movements (-2000 and 0) and with TmcMotionControlStepDir it is working fine for me.

Could you also please check the value of tmc.tmc_mc.current_pos between the movements. It should be -2000*mres ( -4000 if you are using 2 as µstep-resolution).

dcl920108, 2025-03-20:
Hi [Chr157i4n],  

Thank you for your response and for testing the movements on your setup.  

I'm currently using `TmcMotionControlStepDir` as my motion control class. I tried the same test movements (`-2000` and `0`), but I'm still experiencing inconsistent direction changes.  

Here are some of the things I have checked and tested so far:  
- Verified the wiring of the stepper motor (A1, A2, B1, B2).  
- Ensured that `tmc.tmc_mc.current_pos` updates correctly, but I will double-check its value between movements as you suggested.  
- Tried both `RELATIVE` and `ABSOLUTE` movement modes, but the issue persists.  
- Experimented with different `mres` values to see if it affects the movement direction.  
- Checked if any residual movement commands were causing unintended direction changes.  

Since I'm relatively new to stepper motor control, I wonder if I might be missing a key part of the standard control process.  
Are there any best practices or initialization steps I should follow to ensure smooth and consistent movements?  

Thanks again for your help!

dcl920108, 2025-03-20:
Hi [Chr157i4n],  

Here are some of the movement tests we have tried:  

1. **Resetting position before movement:**  
   ```python
   tmc.tmc_mc.current_pos = 0  # Reset current position  
   tmc.run_to_position_fullsteps(-2000, MovementAbsRel.RELATIVE)  
   ```  

2. **Different movement modes:**  
   ```python
   tmc.run_to_position_fullsteps(-2000, MovementAbsRel.RELATIVE)  # Move 2000 fullsteps  
   tmc.reset_position()  # Reset position  
   tmc.run_to_position_fullsteps(-2000, MovementAbsRel.ABSOLUTE)  # Move to -2000 position  
   tmc.run_to_position_fullsteps(0, MovementAbsRel.ABSOLUTE)  # Move back to 0  
   ```  

3. **Testing individual steps:**  
   ```python
   tmc.run_to_position_fullsteps(-1600)  # Move -1600 fullsteps  
   time.sleep(2)  # Wait  
   tmc.run_to_position_fullsteps(-1600, MovementAbsRel.RELATIVE)  # Move another -1600  
   ```  

Despite these tests, the motor still sometimes moves in unexpected directions.  
Could you confirm if there's a specific initialization step we should follow before issuing movements?  

Thanks again for your time!

dcl920108, 2025-03-20:
**Hi,**  

I encountered an issue when trying to use `do_homing()`. The error message I got is:  

```
AttributeError: 'TmcMotionControlVActual' object has no attribute '_tmc_logger'. Did you mean: 'tmc_logger'?
```

It happens in `_tmc_mc_vactual.py` at `set_vactual_dur()`, where `_tmc_logger` is used but seems to be undefined.  

I am using `Tmc2209` with `TmcMotionControlStepDir`, and my setup works fine for normal movements, but `do_homing()` throws this error.  

Is this a bug, or am I missing something in my setup? How should I proceed?  

Thanks in advance!

dcl920108, 2025-03-20:
```python
import time
try:
    from src.tmc_driver.tmc_2209 import *
except ModuleNotFoundError:
    from tmc_driver.tmc_2209 import *

print("\n---")
print("TMC2209 Basic Test Start")
print("---\n")

# -----------------------------
# 1️⃣ Initialize TMC2209
# -----------------------------
tmc = Tmc2209(
    TmcEnableControlPin(21),            # Enable pin EN (GPIO21)
    TmcMotionControlStepDir(16, 20),     # STEP pin (GPIO16), DIR pin (GPIO20)
    TmcComUart("/dev/serial0"),          # UART port
    loglevel=Loglevel.DEBUG              # Enable debug logging
)

tmc.set_motor_enabled(True)             # Enable motor
tmc.set_current(300)                     # Set current to 300mA
tmc.set_microstepping_resolution(4)      # Set microstepping to 1/16 (recommended)
tmc.set_interpolation(True)              # Enable interpolation (smoother motion)
tmc.set_spreadcycle(False)               # Use silent mode (StealthChop)
tmc.set_direction_reg(False)             # Set direction

tmc.acceleration_fullstep = 10000         # Set acceleration
tmc.max_speed_fullstep = 1000             # Set maximum speed

print("\n--- Parameters Set ---\n")

# -----------------------------
# 2️⃣ Execute Homing (Automatic Zeroing)
# -----------------------------
print("Running Homing...")
success = tmc.do_homing(diag_pin=23, revolutions=5, threshold=100)

if success:
    print("Homing successful! Position reset to 0.")
else:
    print("Homing failed! Check sensor or settings.")

# -----------------------------
# 3️⃣ Read Register Status
# -----------------------------
tmc.read_ioin()
tmc.read_chopconf()
tmc.read_drv_status()
tmc.read_gconf()

print("\n--- Registers Read ---\n")

# -----------------------------
# 4️⃣ Perform Test Movement
# -----------------------------
print("Motor Moving Forward...")
tmc.run_to_position_fullsteps(-2000, MovementAbsRel.RELATIVE)  

# -----------------------------
# 5️⃣ Disable Motor & Cleanup
# -----------------------------
tmc.set_motor_enabled(False)             # Disable motor
del tmc                                  # Release resources

print("\n---")
print("TMC2209 Basic Test Finished")
print("---\n")
```

dcl920108, 2025-03-20:
```python
"""
Simple TMC2209 Stepper Motor Test
"""

import time
try:
    from src.tmc_driver.tmc_2209 import *
except ModuleNotFoundError:
    from tmc_driver.tmc_2209 import *

print("\n---")
print("TMC2209 Basic Test Start")
print("---\n")

# -----------------------------
# 1️⃣ Initialize TMC2209
# -----------------------------
tmc = Tmc2209(
    TmcEnableControlPin(21),            # Enable pin EN (GPIO21)
    TmcMotionControlStepDir(16, 20),     # STEP pin (GPIO16), DIR pin (GPIO20)
    TmcComUart("/dev/serial0"),          # UART port
    loglevel=Loglevel.DEBUG              # Enable debug logging
)

# -----------------------------
# 2️⃣ Configure Driver Parameters
# -----------------------------
tmc.set_motor_enabled(True)             # Enable motor
tmc.set_current(300)                     # Set current to 300mA
tmc.set_microstepping_resolution(4)      # Set microstepping to 1/16 (recommended)
tmc.set_interpolation(True)              # Enable interpolation (smoother motion)
tmc.set_spreadcycle(False)               # Use silent mode (StealthChop)
tmc.set_direction_reg(False)             # Set direction

tmc.acceleration_fullstep = 10000         # Set acceleration
tmc.max_speed_fullstep = 1000             # Set maximum speed

print("\n--- Parameters Set ---\n")

# -----------------------------
# 3️⃣ Read and Display Register Status
# -----------------------------
tmc.read_ioin()
tmc.read_chopconf()
tmc.read_drv_status()
tmc.read_gconf()

print("\n--- Registers Read ---\n")

# -----------------------------
# 4️⃣ Perform Test Movements
# -----------------------------

# Reset the current position before movement
tmc.tmc_mc.current_pos = 0  

print("Motor Moving Forward...")
tmc.run_to_position_fullsteps(-2000, MovementAbsRel.RELATIVE)  # Move 2000 full steps forward

time.sleep(2)  # Wait for 2 seconds

print("Motor Moving Backward...")
tmc.run_to_position_fullsteps(2000, MovementAbsRel.RELATIVE)  # Move 2000 full steps backward

time.sleep(2)  # Wait for 2 seconds

print("Motor Moving to Absolute Position...")
tmc.run_to_position_fullsteps(-2000, MovementAbsRel.ABSOLUTE)  # Move directly to -2000 position
time.sleep(2)
tmc.run_to_position_fullsteps(0, MovementAbsRel.ABSOLUTE)  # Return to position 0

print("Movement test completed.")

# -----------------------------
# 5️⃣ Disable Motor & Cleanup
# -----------------------------
tmc.set_motor_enabled(False)             # Disable motor
del tmc                                  # Release resources

print("\n---")
print("TMC2209 Basic Test Finished")
print("---\n")
```

Chr157i4n, 2025-03-20:
No initialization steps should be necessary before a movement.

You could also test with `tmc.tmc_logger.loglevel = Loglevel.MOVEMENT`

in your scripts you have a really high acceleration and max_speed. You should try the same script with lower value, like the ones in the demo scripts

Chr157i4n, 2025-03-20:
this sound to me like a hardware (wiring) issue with the STEP and especially DIR pin

dcl920108, 2025-03-20:
Hi,  
I appreciate your response! However, I don't think it's a wiring issue because:  
1. The motor runs smoothly without noise or stalling.  
2. The issue is that when running the exact same script multiple times, sometimes the motor turns in the expected direction, but other times it reverses.  

I suspect this is related to position accumulation (`current_pos` not resetting properly). Could you confirm if there's a recommended way to reset the position before each movement? Or is there a better approach to ensure direction consistency?  

Thanks for your help!

Chr157i4n, 2025-03-20:
If it behaves differently with several runs of the script, this is definitely not due to the fact that current_pos does not reset itself. Variables are not saved between runs. And it should also not be needed between movement in the same scripts.

There are only 2 ways to change the direction of the motor via the driver:
1. invert the value of the DIR pin (which is done with `set_direction`)
2. invert the register value for shaft in gconf (which is done with `set_direction_reg`)

please test this minimal setup of movement:
```python
tmc.tmc_mc.set_direction(Direction.CW)

for _ in range(2000):
    tmc.tmc_mc.make_a_step()
    time.sleep(0.01)

tmc.tmc_mc.set_direction(Direction.CCW)

for _ in range(2000):
    tmc.tmc_mc.make_a_step()
    time.sleep(0.01)
```

dcl920108, 2025-03-20:
**Hi,**  

I tested the motor using two different scripts, but the motor did not move at all.  

1️⃣ **Single Step Test Script:**  
- The motor was enabled, and I used `make_a_step()` in a loop with a 0.5s delay to observe movement.  
- The script ran without errors, but the motor did not move.  

```python
import time
try:
    from src.tmc_driver.tmc_2209 import *
except ModuleNotFoundError:
    from tmc_driver.tmc_2209 import *

print("\n---")
print("TMC2209 Single Step Test")
print("---\n")

# Initialize the TMC2209 driver
tmc = Tmc2209(
    TmcEnableControlPin(21),            
    TmcMotionControlStepDir(16, 20),     
    TmcComUart("/dev/serial0"),          
    loglevel=Loglevel.DEBUG              
)

tmc.set_motor_enabled(True)  # Enable the motor
tmc.set_current(500)  # Increase current to ensure motor gets enough power
tmc.set_microstepping_resolution(4)  # Lower microstepping to make motion clearer

print("Stepping motor forward...")
for _ in range(10):  # Only 10 steps
    tmc.tmc_mc.make_a_step()
    time.sleep(0.5)  # Wait between steps to observe motion

tmc.set_motor_enabled(False)  # Disable the motor
del tmc  # Cleanup

print("\n---")
print("TMC2209 Single Step Test Finished")
print("---\n")
```

2️⃣ **Minimal Direction Test Script:**  
- The motor was set to move **2000 steps in CW direction**, then **2000 steps in CCW direction**.  
- The script executed without errors, but the motor showed no motion.  

```python
import time
try:
    from src.tmc_driver.tmc_2209 import *
except ModuleNotFoundError:
    from tmc_driver.tmc_2209 import *

print("\n---")
print("TMC2209 Minimal Direction Test")
print("---\n")

# Initialize the TMC2209 driver
tmc = Tmc2209(
    TmcEnableControlPin(21),            
    TmcMotionControlStepDir(16, 20),     
    TmcComUart("/dev/serial0"),          
    loglevel=Loglevel.DEBUG              
)

tmc.set_motor_enabled(True)  # Enable the motor

# Test CW direction
print("Moving CW...")
tmc.tmc_mc.set_direction(Direction.CW)
for _ in range(2000):
    tmc.tmc_mc.make_a_step()
    time.sleep(0.01)

time.sleep(1)  # Pause before switching direction

# Test CCW direction
print("Moving CCW...")
tmc.tmc_mc.set_direction(Direction.CCW)
for _ in range(2000):
    tmc.tmc_mc.make_a_step()
    time.sleep(0.01)

tmc.set_motor_enabled(False)  # Disable the motor
del tmc  # Cleanup

print("\n---")
print("TMC2209 Minimal Direction Test Finished")
print("---\n")
```

I am new to this, and unfortunately, no one in my company can assist me with debugging this issue. I am confused about what might be wrong. Could you suggest what I should check next?  

Thanks in advance!

Chr157i4n, 2025-03-20:
hmmmm. I don't really know why that might be. My next step would be to measure the STEP and DIR pins on the driver with an oscilloscope:
DIR should be high or low, depending on the direction of the movement
STEP should be pulsed for at least 1µs per step.

If these signals arrive correctly, the problem is probably with the driver itself, otherwise it may be with the cabling between the driver and Pi.

Could you please post a picture of your setup and wiring.

dcl920108, 2025-03-21:
**Hi [Chr157i4n],**  

Thanks for your continuous help and support! I really appreciate it.  

I will test the STEP and DIR signals with an oscilloscope tomorrow and check if they are behaving as expected. My wiring setup follows your design exactly, so I will focus on verifying the signals first.  

I’ll update you once I have the results. Thanks again!  

[image]

dcl920108, 2025-03-24:
Hi Chr157i4n,

Thank you so much for your continuous support!

Everything is working perfectly now. The issue was caused by incorrect wiring between the stepper motor and the TMC2209 driver. Specifically, the A1/A2 and B1/B2 motor coil pairs were not connected to the correct driver outputs. After checking the coil pairs with a multimeter and correcting the connections, the motor now runs smoothly and responds correctly to direction and step commands.

Really appreciate your guidance throughout the process!

Best regards,

Chr157i4n, 2025-03-24:
I'm glad to hear that.

closed 2025-03-24 by Chr157i4n