Chr157i4n/PyTmcStepper #113: Incorrect value mask for vactual mangles negative step values

jamesratzlaff, 2025-05-13:
**Describe the bug**
[Incorrect value mask for vactual](https://github.com/Chr157i4n/PyTmcStepper/blob/ee0afd5ef20d99209f231164dee6f5e178f066c7/src/tmc_driver/reg/_tmc220x_reg.py#L147)

**To Reproduce**
modified demo_05 to run

```python
tmc.run_to_position_steps(100,movement_abs_rel=MovementAbsRel.RELATIVE)
time.sleep(1)
tmc.run_to_position_steps(-100,movement_abs_rel=MovementAbsRel.RELATIVE)
```
The forward motion works fine, the backwards (-100 steps) motion barely does anything.

The values vactual is supposed to accept is a signed integer and I believe the mask that you have is chopping off the sign bit. I changed the mask to 0xFFFFFF and it works as expected.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Versions**
- Raspberry Pi 4B
- Raspberry Pi OS Lite 6.12.25+rpt-rpi-v8
- v0.7.5
- BTT TMC2209 v1.2

Chr157i4n, 2025-05-13:
yes, you are totally right.

The VActual register uses 24 bits and should therefore have 0xFFFFFFFF as the bit mask.

[image]

If you want, you can create a pull request for fixing it 😁, otherwise i will fix it at the weekend.

jamesratzlaff, 2025-05-14:
I created the pull request, though I just noticed you suggested 0xFFFFFFFF (32-bits) as the mask instead of 0xFFFFFF (24-bits).  For sanity's sake, I referenced the [manufacturer's (analogdevices) implementation and they are using 0x00FFFFFF](https://github.com/analogdevicesinc/PyTrinamic/blob/a8c8acd9a9f51f182a30dc3e33929e8fcc5b018c/pytrinamic/ic/TMC2209.py#L137) as the mask.

Chr157i4n, 2025-05-14:
yes, sorry I thought I had copied your value...